john30/ebusd #1512: Home Assistant 2025.3.1:  Entity missing; Error 'The unit of measurement `%` is not valid together with device class `temperature`'

jan666, 2025-03-11:
### Description

I updated to Home Assistant 2025.3.1 and now I miss the Mixer Movemet of my VR 70. 

`2025-03-11 07:22:07.351 ERROR (MainThread) [homeassistant.components.mqtt.entity] Error 'The unit of measurement `%` is not valid together with device class `temperature`' when processing MQTT discovery message topic: 'homeassistant/sensor/ebusd_vr_70_Mc2FlowTempDesired_MixerMovement/config', message: '{'unique_id': 'ebusd_vr_70_Mc2FlowTempDesired_MixerMovement', 'device': {'identifiers': 'ebusd_vr_70', 'manufacturer': 'ebusd.eu', 'via_device': 'ebusd', 'sw_version': '24.1', 'suggested_area': 'Heating', 'name': 'ebusd vr_70'}, 'value_template': '{{value_json["MixerMovement"].value}}', 'state_topic': 'ebusd/vr_70/Mc2FlowTempDesired', 'unit_of_measurement': '%', 'state_class': 'measurement', 'device_class': 'temperature', 'name': 'Mc2FlowTempDesired MixerMovement'}'`

I think I saw this error before (everything worked with 2025.3.0) but it was more a warning?

There is more

```
2025-03-11 07:21:51.213 ERROR (MainThread) [homeassistant.components.mqtt.entity] Error 'The unit of measurement `percent` is not valid together with device class `power`' when processing MQTT discovery message topic: 'homeassistant/sensor/ebusd_hmu_BuildingCircuitPumpPower_percent3/config', message: '{'unique_id': 'ebusd_hmu_BuildingCircuitPumpPower_percent3', 'device': {'identifiers': 'ebusd_hmu', 'manufacturer': 'ebusd.eu', 'via_device': 'ebusd', 'sw_version': '24.1', 'suggested_area': 'Heating', 'name': 'ebusd hmu'}, 'value_template': '{{value_json["percent3"].value}}', 'state_topic': 'ebusd/hmu/BuildingCircuitPumpPower', 'unit_of_measurement': 'percent', 'state_class': 'measurement', 'device_class': 'power', 'name': 'BuildingCircuitPumpPower '}'
2025-03-11 07:21:51.217 ERROR (MainThread) [homeassistant.components.mqtt.entity] Error 'The unit of measurement ` ` is not valid together with device class `energy`' when processing MQTT discovery message topic: 'homeassistant/sensor/ebusd_hmu_CopCoolingMonthSensor_energy/config', message: '{'unique_id': 'ebusd_hmu_CopCoolingMonthSensor_energy', 'device': {'identifiers': 'ebusd_hmu', 'manufacturer': 'ebusd.eu', 'via_device': 'ebusd', 'sw_version': '24.1', 'suggested_area': 'Heating', 'name': 'ebusd hmu'}, 'value_template': '{{value_json["energy"].value}}', 'state_topic': 'ebusd/hmu/CopCoolingMonthSensor', 'unit_of_measurement': ' ', 'state_class': 'total_increasing', 'device_class': 'energy', 'name': 'CopCoolingMonthSensor '}'
2025-03-11 07:21:51.218 ERROR (MainThread) [homeassistant.components.mqtt.entity] Error 'The unit of measurement ` ` is not valid together with device class `energy`' when processing MQTT discovery message topic: 'homeassistant/sensor/ebusd_hmu_CopCoolingSensor_energy/config', message: '{'unique_id': 'ebusd_hmu_CopCoolingSensor_energy', 'device': {'identifiers': 'ebusd_hmu', 'manufacturer': 'ebusd.eu', 'via_device': 'ebusd', 'sw_version': '24.1', 'suggested_area': 'Heating', 'name': 'ebusd hmu'}, 'value_template': '{{value_json["energy"].value}}', 'state_topic': 'ebusd/hmu/CopCoolingSensor', 'unit_of_measurement': ' ', 'state_class': 'total_increasing', 'device_class': 'energy', 'name': 'CopCoolingSensor '}'

```

### Actual behavior

Entity missing

### Expected behavior

Entity should be there again. MixerMovement with unit % should not be a temperature?

### ebusd version

24.1

### ebusd arguments

--scanconfig -d ens:/dev/ttyACM0 -c /home/pi/git/ebusd-configuration/ebusd-2.1.x/de/ --log=all:error --httpport=8080 --mqtthost 192.168.42.254 --mqttport 1883 --mqttuser ebusd --mqttpass ebusd --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqttjson

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via USB

### Logs

nothing relevant

chrizzzp, 2025-03-11:
Same here after update to HA core 2025.3.1. More entities not getting updated any longer. Seems to affect for instance pressure entities recognized as 'temperature' device class.

- Error 'The unit of measurement **`'bar'`** is not valid together with device class **`'temperature'`** when processing MQTT discovery message topic: 'homeassistant/sensor/**ebusd_hmu_Status07_DisplaySystemPressure**/config', message: '{'unique_id': 'ebusd_hmu_Status07_DisplaySystemPressure', 'device': {'identifiers': 'ebusd_hmu', 'manufacturer': 'ebusd.eu', 'via_device': 'ebusd', 'sw_version': '24.1', 'suggested_area': 'Heating', 'name': 'ebusd hmu'}, 'value_template': '{{value_json["DisplaySystemPressure"].value}}', 'state_topic': 'ebusd/hmu/Status07', 'unit_of_measurement': 'bar', 'state_class': 'measurement', 'device_class': 'temperature', 'name': 'Status07 DisplaySystemPressure'}'

- Error 'The unit of measurement **`'bar'`** is not valid together with device class **`'temperature`'** when processing MQTT discovery message topic: 'homeassistant/sensor/**ebusd_vwz00_RunDataSystemPressure_pressv**/config', message: '{'unique_id': 'ebusd_vwz00_RunDataSystemPressure_pressv', 'device': {'identifiers': 'ebusd_vwz00', 'manufacturer': 'ebusd.eu', 'via_device': 'ebusd', 'sw_version': '24.1', 'suggested_area': 'Heating', 'name': 'ebusd vwz00'}, 'value_template': '{{value_json["pressv"].value}}', 'state_topic': 'ebusd/vwz00/RunDataSystemPressure', 'unit_of_measurement': 'bar', 'state_class': 'measurement', 'device_class': 'temperature', 'name': 'RunDataSystemPressure pressv'}'

- Error 'The unit of measurement **`'bar'`** is not valid together with device class **`'temperature'`** when processing MQTT discovery message topic: 'homeassistant/sensor/**ebusd_vwz00_Status07_DisplaySystemPressure**/config', message: '{'unique_id': 'ebusd_vwz00_Status07_DisplaySystemPressure', 'device': {'identifiers': 'ebusd_vwz00', 'manufacturer': 'ebusd.eu', 'via_device': 'ebusd', 'sw_version': '24.1', 'suggested_area': 'Heating', 'name': 'ebusd vwz00'}, 'value_template': '{{value_json["DisplaySystemPressure"].value}}', 'state_topic': 'ebusd/vwz00/Status07', 'unit_of_measurement': 'bar', 'state_class': 'measurement', 'device_class': 'temperature', 'name': 'Status07 DisplaySystemPressure'}'

itsopti, 2025-03-11:
Have a look at https://github.com/home-assistant/core/issues/140313

Should be fixed at next 2025.3.2 but ebusd needs to use correct units of measurement per device class until 2025.07

jan666, 2025-03-12:
Yes, it works again with 2025.3.2 (WARNING, not ERROR again)

`2025-03-12 06:43:58.660 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.vr_70_mc2flowtempdesired_mixermovement (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('temperature') it is using; expected one of ['°C', 'K', '°F']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22`

but like @itsopti said it will stop working in 2025.7

chrizzzp, 2025-03-12:
> Should be fixed at next 2025.3.2 but ebusd needs to use correct units of measurement per device class until 2025.07

I think in case of the mixer movement the unit of measurement is OK, but the device class should not be 'temperature', but rather 'power factor' which allows the unit '%'. Or how to deal with sensors which do not fit in any of [these](https://developers.home-assistant.io/docs/core/entity/sensor/) device class categories?

Same with the pressure entities above: the device class is misassigned to 'temperature' instead of 'pressure'.

So is this an issue of the mqtt_hassio.cfg (which I did not update with 24.1 as I have custom changes) or of ebusd (24.1) itself?

jan666, 2025-04-07:
anyone with an idea how to fix this? 2025.7 is getting near...

chrizzzp, 2025-04-07:
@jan666 
Which version of the` mqtt_hassio.cfg` are you using? The one that came with ebusd 24.1 or an own (previously modified) one?  Maybe editing `mqtt_hassio.cfg` is necessary? I think the entities are classified there. I haven't throroughly looked into this, though...

jan666, 2025-04-07:
> [@jan666](https://github.com/jan666) Which version of the` mqtt_hassio.cfg` are you using? The one that came with ebusd 24.1 or an own (previously modified) one? 

I'm using the version from the contrib folder, edited to get more info:

`< filter-name = status|temp|humidity|yield|count|energy|power|runtime|hours|starts|mode|curve|^load$|^party$|sensor|timer|actor|mixer`

and to be able to write:

`< filter-direction = r|u|^w`

> Maybe editing `mqtt_hassio.cfg` is necessary? I think the entities are classified there. I haven't throroughly looked into this, though...

Maybe, I already looked into the file before opening the issue. But I dont really undertand whats going on there

john30, 2025-04-13:
this needs to be fixed in the ebusd-configuration files as e.g. the message with name "Mc1FlowTempDesired" is misleading as it contains way more than just the desired temperature. should rather be "status"/"mode"/"control" or something. also the field names are ugly with "FT" prefix which won't resolve to a temperature.
so feel free to open a PR on https://github.com/john30/ebusd-configuration/

john30, 2025-04-13:
> Same here after update to HA core 2025.3.1. More entities not getting updated any longer. Seems to affect for instance pressure entities recognized as 'temperature' device class.
>
>     * Error 'The [...]

that is most probably the result of an incorrect mqtt-hassio.cfg or is this original and also the csvs?

closed 2025-04-13 by john30

jan666, 2025-04-14:
@john30 I'm using the "stock" vr 70 csv, just replaced the 'w' parts with 'uw':

```
# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment,field2,part (m/s),datatypes/templates,divider/values,unit,comment,field3,part (m/s),datatypes/templates,divider/values,unit,comment,field4,part (m/s),datatypes/templates,divider/values,unit,comment,field5,part (m/s),datatypes/templates,divider/values,unit,comment,field6,part (m/s),datatypes/templates,divider/values,unit,comment,field7,part (m/s),datatypes/templates,divider/values,unit,comment
#,VR_70,VR 70,109 109,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*r,,,,,,"B523",,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*uw,,,,,,"B523",,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
uw,,SetActorState,,,,,"01",R1,,UCH,0=off;20=on,,,R2,,UCH,0=off;20=on,,,R3,,UCH,0=off;20=on,,,R4,,UCH,0=off;20=on,,,R5,,UCH,0=off;20=on,,,R6,,UCH,0=off;20=on,,,S7,,UCH,0=off;20=on,,
uw,,Mc1FlowTempDesired,,,,,"0200",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
uw,,Mc2FlowTempDesired,,,,,"0201",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
r,,SensorData,,,,,"03",S1,,temp,,,,S2,,temp,,,,S3,,temp,,,,S4,,temp,,,,S5,,temp,,,,S6,,temp,,,,ignore,,HEX:3,,,
!include,errors.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,

```

So IMHO I did nothing wrong, but the mixer movement in % is still temperature. Can you help me find out how to fix this?

john30, 2025-04-24:
if you adjusted it already, then try using e.g. Mc1FlowStatus instead of Mc1FlowTempDesired and the same for the second one

jan666, 2025-07-03:
```
diff --git a/ebusd-2.1.x/de/vaillant/52.vr_70.csv b/ebusd-2.1.x/de/vaillant/52.vr_70.csv
index cbd485c..5482c01 100644
--- a/ebusd-2.1.x/de/vaillant/52.vr_70.csv
+++ b/ebusd-2.1.x/de/vaillant/52.vr_70.csv
@@ -3,7 +3,7 @@
 *r,,,,,,"B523",,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
 *uw,,,,,,"B523",,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
 uw,,SetActorState,,,,,"01",R1,,UCH,0=off;20=on,,,R2,,UCH,0=off;20=on,,,R3,,UCH,0=off;20=on,,,R4,,UCH,0=off;20=on,,,R5,,UCH,0=off;20=on,,,R6,,UCH,0=off;20=on,,,S7,,UCH,0=off;20=on,,
-uw,,Mc1FlowTempDesired,,,,,"0200",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
-uw,,Mc2FlowTempDesired,,,,,"0201",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
+uw,,Mc1FlowStatus,,,,,"0200",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
+uw,,Mc2FlowStatus,,,,,"0201",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
 r,,SensorData,,,,,"03",S1,,temp,,,,S2,,temp,,,,S3,,temp,,,,S4,,temp,,,,S5,,temp,,,,S6,,temp,,,,ignore,,HEX:3,,,
 !include,errors.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,

```

did the trick. thank you!

pascalwinters, 2025-07-21:
> ```
> diff --git a/ebusd-2.1.x/de/vaillant/52.vr_70.csv b/ebusd-2.1.x/de/vaillant/52.vr_70.csv
> index cbd485c..5482c01 100644
> --- a/ebusd-2.1.x/de/vaillant/52.vr_70.csv
> +++ [...]

This is also working fine for me, except this error. Do you have it also? It's related to the FTStatus:
```
"FTStatus": {
    "value": 2
  }
```
I receive this error in HomeAssistant logging:
`Ignoring invalid option received on topic 'ebusd/vr_71/Mc1FlowStatus', got '2', allowed: off, on`