john30/ebusd #1520: MQTT Explorer shows more than MQTT Mosquitto at Home Assistant

Buggy1987, 2025-03-22:
### Description

Hello I use the ebusd dongle from elecrow. It gets many sensors from my wolf cwl 300 air pump. 

But the home assistant MQTT broker mosquito get not all values I can see in the MQTT explorer. 

And how can I set values with the ebusd?

### Actual behavior

No errors I can see

### Expected behavior

Sensors missing 

### ebusd version

current source from git

### ebusd arguments

/etc/ebusd/mqtt-hassio.cfg

### Operating system

other

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

[image]
[image]

Koky05, 2025-03-24:
All depends on your filters and sensor definitions in mqtt-hassio.cfg file you use.
For example I comment out all filters and change direction to get all messages, also add some mode sensor definition to better sensor mapping and now I have 1200 sensors provided from eBus Deamon into HA.

Buggy1987, 2025-03-24:
Ah ok thank you. Short question: How can i access the mqtt-hassio.cfg? I already installed the file explorer but he doesnt shows me

Koky05, 2025-03-24:
In  mqtt-hassio.cfg is several filters:
```
...
# HA integration: filtering only seen messages to avoid unnecessary "pollution" and auto-poll all matched messages
filter-seen = 1
# include only messages having a priority less than or equal to the specified value.
filter-priority =
...
# HA integration: filter to some useful names for monitoring the heating circuit
filter-name = status|temp|yield|fuel|count|energy|power|runtime|hours|starts|mode|curve|load|party|cooling|heat|sensor|data|error|pressure|flame|date|time|a|e|i|y|o|u|day
...
# HA integration: writable messages excluded for now
filter-direction = r|u|^w
...
# HA integration: exclude the useless sensor ok/cutoff info from temperature sensor.
filter-non-field = ^sensor$
...
```

If you put `#` in fron of that lines you comment out all filters and you got more values into HA. Or you can modify them in way you need it. Filter by part of name, direction, priority, sensor type, etc.

Also there is some senor type definitions:
```
...
# This is also an implicit field type filter as missing or empty mappings are not published at all.
type_map-number = number
type_map-list = string
# HA integration: skip string/date/time types completely
type_map-string = string
type_map-date = date
type_map-time = time
type_map-datetime = datetime
...
type_switch-w-number =
    number,heating curve, = curve*%%
    number,heating curve, = Curve*%%
    number,temperature, = temp|,°C$|,K$
    number,power_factor, = power*%%
    number,power, = power|,kW$|,W$
    number,voltage, = volt|,V$
    number,current, = current,|currentvalue,|,A$
    number,pressure, = bar$
    number,gas, = gas*/min$
    number,volume, = m³$
    number,volume_flow_rate, = m³/h$
    number,humidity, = humid*%%$
    number,, = poweron|count,|countvalue
    number,, = integral|,°min$
    number,, = air|,m³/h$
    number,, = curve,|curvevalue,
    number,, = 

# HA integration: the mapping list for numeric sensor entities by field type, name, message, and unit.
type_switch-number =
    sensor,heating curve,measurement = curve
    sensor,heating curve,measurement = Curve
    sensor,power_factor,measurement = power*%%
    sensor,temperature,measurement = temp|,°C$|,K$
    sensor,power,measurement = power|,kW$|,W$
    sensor,voltage,measurement = volt|,V$
    sensor,current,measurement = current,|currentvalue,|,A$
    sensor,energy,total_increasing = energy|,Wh$
    sensor,yield,total_increasing = total*,Wh$
    sensor,duration,total_increasing = hours|days|,h$
    sensor,pressure,measurement = bar$|Pa$
    sensor,gas,measurement = gas*/min$
    sensor,volume,total_increasing = total*,m³$
    sensor,volume,measurement = m³$
    sensor,volume_flow_rate,measurement = m³/h$
    sensor,carbon_dioxide,measurement = CO2*,ppm$
    sensor,humidity,measurement = humid*%%$
    sensor,,total_increasing = poweron|count,|countvalue
    sensor,,measurement = integral|,°min$
    sensor,,total_increasing = hours|,h$
    sensor,,total_increasing = starts*,$
    sensor,,measurement = air|,m³/h$
    sensor,, = 

# HA integration: the mapping list for (potentially) writable binary switch entities by field type, name, message, and unit.
type_switch-w-list =
    switch,, = onoff
    switch,,,yesno = yesno
    select,, =

# HA integration: the mapping list for rather binary sensor entities by field type, name, message, and unit.
type_switch-list =
    binary_sensor,,measurement = onoff
    binary_sensor,,measurement,yesno = yesno
    binary_sensor,,name = Flame
    sensor,,,list =

# HA integration: the mapping list for (potentially) writable string entities containing a time value by field type, name, message, and unit.
type_switch-w-time =
    text,,,time = from,|to,|time2,|timer
    text,,,time3 = time,|timevalue,

type_switch-time =
    sensor,,,time = from,|to,|time2,|timer
    sensor,,,time3 = time,|timevalue,

# HA integration: currently unused mapping lists for non-numeric/non-binary entities.
type_switch-w-date =
    text,,,date = date,
    text,, = date,
    text,, =

type_switch-w-datetime =
    text,, = btime,
    text,, = dcfstate,
    text,, = datetime,
    text,, = bdate,
    text,, = 

type_switch-w-string =
    text,, = date,
    text,, = datetime,
    text,, = dcfstate,
    text,, = bdate,
    text,, = btime,
    text,, = phone,
    text,, =

type_switch-date =
    sensor,, = date,
    sensor,, =

type_switch-datetime =
    sensor,, =

type_switch-string =
    sensor,, =
...
type_part-binary_sensor = ,
  "payload_on":"on",
  "payload_off":"off"%state_class

type_part-binary_sensoryesno = ,
  "payload_on":"yes",
  "payload_off":"no"%state_class

# HA integration: %type_part variable for text %type_topic
type_part-texttime = ,
  "command_topic":"%topic/set",
  "pattern": "^[012][0-9]:[0-5][0-9]$"%state_class

type_part-textphone = ,
  "command_topic":"%topic/set",
  "pattern": "^[0-9]{9}$"%state_class

type_part-texttime3 = ,
  "command_topic":"%topic/set",
  "pattern": "^[012][0-9]:[0-5][0-9]:[0-5][0-9]$"%state_class

type_part-textdate = ,
  "command_topic":"%topic/set",
  "pattern": "^[0-3][0-9].[01][0-9].20[0-3][0-9]$"%state_class

type_part-textdatetime = ,
  "command_topic":"%topic/set",
  "pattern": "^[0-3][0-9].[01][0-9].20[0-3][0-9] [012][0-9]:[0-5][0-9]$"%state_class

type_part-text = ,
  "command_topic":"%topic/set"%state_class
...
```

Please take that values as example because I modify my files according my need.

john30, 2025-04-13:
closed as a matter of configuration

closed 2025-04-13 by john30