john30/ebusd #1540: Document compilation on macOS

wimleers, 2025-05-11:
### Description

I've been using `ebusd` via Docker for ~2 years. But stability is poor. Not due to `ebusd`'s fault, but due to Docker. Docker Desktop for macOS is horribly unreliable — it used to be better, but in the past ~year, Docker Desktop has clearly gotten worse.
So I'd like to get rid of it, and just run `ebusd` natively.

A few years after #673, version `25.1` claims improved macOS compatibility in the changelog. Wonderful! 🙏😊

### Actual behavior

Unfortunately, this is what happens:
```
$ git clone https://github.com/john30/ebusd.git
Cloning into 'ebusd'...
remote: Enumerating objects: 19159, done.
remote: Counting objects: 100% (1109/1109), done.
remote: Compressing objects: 100% (193/193), done.
remote: Total 19159 (delta 973), reused 931 (delta 916), pack-reused 18050 (from 2)
Receiving objects: 100% (19159/19159), 10.37 MiB | 30.97 MiB/s, done.
Resolving deltas: 100% (12020/12020), done.
admin at mist in ~/Desktop
$ cd ebusd
admin at mist in ~/Desktop/ebusd on master*
$ git checkout 25.1
Note: switching to '25.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at f1818cfa updated version to 25.1
admin at mist in ~/Desktop/ebusd on (HEAD detached at 25.1)*
$ cmake .
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- The C compiler identification is AppleClang 16.0.0.16000026
-- The CXX compiler identification is AppleClang 16.0.0.16000026
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for arpa/inet.h
-- Looking for arpa/inet.h - found
-- Looking for dirent.h
-- Looking for dirent.h - found
-- Looking for fcntl.h
-- Looking for fcntl.h - found
-- Looking for netdb.h
[... 262 lines cut ...]
n/a

### Operating system

other

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via USB

### Related integration

other

### Logs

N/A

OLFDB, 2025-05-12:
There are several compatibility issues. 

You need to rename the file VERSION to VERSION_FILE.

Edit CMakeLists.txt 

file(STRINGS "VERSION" VERSION)

to

file(STRINGS "VERSION_FILE" VERSION)

and add

if(APPLE)
   add_compile_definitions(APPLE)
endif()

In src/lib/knx/knxnet.h and src/lib/utils/tcpsocket.h change the condition for #include <machine/endian.h> to 
#if defined(__FreeBSD__) || defined(APPLE)

In src/lib/utils/tcpsocket.cpp change the condition for #include <poll.h> to 
#if defined(__FreeBSD__) || defined(APPLE)

Now you can compile using make but linking will fail with 

ld: library 'rt' not found

edit src/ebusd/CMakeFiles/ebusd.dir/link.txt and remove -lrt

I tried to remove -lrt by editing CMakeLists.txt and changing

set(CMAKE_REQUIRED_LIBRARIES pthread rt)

to 

if(NOT APPLE)   
set(CMAKE_REQUIRED_LIBRARIES pthread rt)
else()
set(CMAKE_REQUIRED_LIBRARIES pthread)
endif()

but that is not removing -l rt in src/ebusd/CMakeFiles/ebusd.dir/link.txt

wimleers, 2025-10-24:
If I follow the instructions provided by @OLFDB (thanks! ❤️), I am now down to:
```
$ make
[  2%] Building CXX object src/lib/knx/CMakeFiles/knx.dir/knx.cpp.o
In file included from /Users/admin/Desktop/ebusd/src/lib/knx/knx.cpp:30:
/Users/admin/Desktop/ebusd/src/lib/knx/knxnet.h:481:32: warning: implicit conversion changes signedness: 'unsigned long' to 'ssize_t' (aka 'long') [-Wsign-conversion]
  481 |     ssize_t dataLen = totalLen - (lOffset+sizeof(knxnet_l_data_header_t));
      |             ~~~~~~~   ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/admin/Desktop/ebusd/src/lib/knx/knxnet.h:535:21: warning: implicit conversion changes signedness: 'int' to 'size_t' (aka 'unsigned long') [-Wsign-conversion]
  535 |     memcpy(data, d, *recvlen);  // including the TPCI/APCI octet 6
      |     ~~~~~~          ^~~~~~~~
/Users/admin/Desktop/ebusd/src/lib/knx/knxnet.h:534:33: warning: implicit conversion loses integer precision: 'ssize_t' (aka 'long') to 'int' [-Wshorten-64-to-32]
  534 |     *recvlen = size > dataLen ? dataLen : size;
      |              ~                  ^~~~~~~
/Users/admin/Desktop/ebusd/src/lib/knx/knxnet.h:574:107: warning: implicit conversion changes signedness: 'int' to 'unsigned long' [-Wsign-conversion]
  574 |     size_t totalLen = sizeof(knxnet_header_t)+sizeof(knxnet_cemi_header_t)+sizeof(knxnet_l_data_header_t)+len;
      |                                                                                                          ~^~~
/Users/admin/Desktop/ebusd/src/lib/knx/knxnet.h:592:31: warning: implicit conversion loses integer precision: 'int' to 'uint8_t' (aka 'unsigned char') [-Wimplicit-int-conversion]
  592 |     l->informationLength = len-1;  // subtracting the TPCI/APCI
      |                          ~ ~~~^~
/Users/admin/Desktop/ebusd/src/lib/knx/knxnet.h:595:21: warning: implicit conversion changes signedness: 'int' to 'size_t' (aka 'unsigned long') [-Wsign-conversion]
  595 |     memcpy(d, data, len);
      |     ~~~~~~          ^~~
/Users/admin/Desktop/ebusd/src/lib/knx/knxnet.h:603:12: warning: implicit conversion loses integer precision: 'int' to 'uint8_t' (aka 'unsigned char') [-Wimplicit-int-conversion]
  603 |     d[0] = tpci;
      |          ~ ^~~~
7 warnings generated.
[  4%] Linking CXX static library libknx.a
[  4%] Built target knx
[  6%] Building CXX object src/lib/utils/CMakeFiles/utils.dir/arg.cpp.o
/Users/admin/Desktop/ebusd/src/lib/utils/arg.cpp:102:57: warning: implicit conversion changes signedness: 'int' to 'size_t' (aka 'unsigned long') [-Wsign-conversion]
  102 |   struct option *longOpts = (struct option*)calloc(count+1, sizeof(struct option));  // room for EOF
      |                                             ~~~~~~ ~~~~~^~
/Users/admin/Desktop/ebusd/src/lib/utils/arg.cpp:103:68: warning: implicit conversion changes signedness: 'int' to 'size_t' (aka 'unsigned long') [-Wsign-conversion]
  103 |   char *shortChars = reinterpret_cast<char*>(calloc(shortCharsCount+1, sizeof(char)));  // room for \0
      |                                              ~~~~~~ ~~~~~~~~~~~~~~~^~
/Users/admin/Desktop/ebusd/src/lib/utils/arg.cpp:104:53: warning: implicit conversion changes signedness: 'int' to 'size_t' (aka 'unsigned long') [-Wsign-conversion]
  104 |   int *shortIndexes = reinterpret_cast<int*>(calloc(shortCharsCount, sizeof(int)));
      |                                              ~~~~~~ ^~~~~~~~~~~~~~~
/Users/admin/Desktop/ebusd/src/lib/utils/arg.cpp:105:68: warning: implicit conversion changes signedness: 'int' to 'size_t' (aka 'unsigned long') [-Wsign-conversion]
  105 |   char *shortOpts = reinterpret_cast<char*>(calloc(2+shortOptsCount+1, sizeof(char)));  // room for +, :, and \0
      |                                             ~~~~~~ ~~~~~~~~~~~~~~~~^~
/Users/admin/Desktop/ebusd/src/lib/utils/arg.cpp:254:17: warning: implicit conversion changes signedness: 'long' to 'size_t' (aka 'unsigned long') [-Wsign-conversion]
  254 |       cnt = eol - str;
      |           ~ ~~~~^~~~~
/Users/admin/Desktop/ebusd/src/lib/utils/arg.cpp:256:17: warning: implicit conversion changes signedness: 'long' to 'size_t' (aka 'unsigned long') [-Wsign-conversion]
  256 |       cnt = end - str;
      |           ~ ~~~~^~~~~
6 warnings generated.
[  9%] Building CXX object src/lib/utils/CMakeFiles/utils.dir/log.cpp.o
[ 11%] Building CXX object src/lib/utils/CMakeFiles/utils.dir/tcpsocket.cpp.o
/Users/admin/Desktop/ebusd/src/lib/utils/tcpsocket.cpp:57:44: warning: implicit conversion changes signedness: 'int' to 'size_t' (aka 'unsigned long') [-Wsign-conversion]
   57 |   memcpy(sin_addr, he->h_addr_list[0], he->h_length);
      |   ~~~~~~                               ~~~~^~~~~~~~
/Users/admin/Desktop/ebusd/src/lib/utils/tcpsocket.cpp:196:29: error: use of undeclared identifier 'POLLIN'
  196 |       ret = socketPoll(sfd, POLLIN|POLLOUT, tcpConnToUdpOptions);
      |                             ^
/Users/admin/Desktop/ebusd/src/lib/utils/tcpsocket.cpp:196:36: error: use of undeclared identifier 'POLLOUT'
  196 |       ret = socketPoll(sfd, POLLIN|POLLOUT, tcpConnToUdpOptions);
      |                                    ^
/Users/admin/Desktop/ebusd/src/lib/utils/tcpsocket.cpp:238:15: error: use of undeclared identifier 'POLLIN'
  238 |   if (which & POLLIN) {
      |               ^
/Users/admin/Desktop/ebusd/src/lib/utils/tcpsocket.cpp:241:15: error: use of undeclared identifier 'POLLOUT'
  241 |   if (which & POLLOUT) {
[... cut ...]
  477 |     ret = socketPoll(sock, POLLIN, 1);
      |                            ^
/Users/admin/Desktop/ebusd/src/lib/utils/tcpsocket.cpp:479:25: error: use of undeclared identifier 'POLLIN'
  479 |     if (ret > 0 && (ret&POLLIN)) {
      |                         ^
2 warnings and 8 errors generated.
make[2]: *** [src/lib/utils/CMakeFiles/utils.dir/tcpsocket.cpp.o] Error 1
make[1]: *** [src/lib/utils/CMakeFiles/utils.dir/all] Error 2
make: *** [all] Error 2
admin at mist in ~/Desktop/ebusd on macos_build*
```

wimleers, 2025-10-24:
OMG only now I found https://github.com/john30/ebusd/pull/1542! Testing that…