john30/ebusd #1563: Arbitration lost after ebusd update

hce, 2025-07-07:
### Description

I set up ebusd with the then current esera ebus converter ethernet version and a wolf heating system on April 2021. I've manually compiled and deployed the then current version of ebusd. This setup has been running until last week with few problems. I was able to read out most parameters passively, and I actively read two parameters actively every five minutes. I've plotted munin graphs of most parameters. Everything worked fine. The ebusd ran on a linux VM on a FreeBSD host and there were three managed L2 switches between the ebusd VM and the ebus converter. No problems at all with the latency.

Last week, I noticed by chance that ebusd crashed once a week. Since my logs are only temporary ones I cannot say if this started to happen a week before that or if it has been going on for years.

Anyway, I have upgraded to the latest ebusd version. Since then I cannot synchronize the bus anymore. I changed ebusd to go to readonly mode and am still able to read out all parameters passively, works just as well as before. But I cannot send on the ebus anymore. I have set up a new machine on an intel system, no virtualization, 1.5 meter long ethernet cable, directly connected to the esera ebusd converter. This didn't change anything.

### Actual behavior

I am only able to use the ebus in read only mode with the latest versions of ebusd.

### Expected behavior

Read write usage of the ebus just like with the 2021 build of ebusd

### ebusd version

current source from git

### ebusd arguments

--updatecheck=off -d udp:192.168.7.225:5000 -c /etc/ebusd/wolf -f -r --loglevel=error

### Operating system

other

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter v3 Ethernet

### Logs

root@tb /h/hc# sudo -u ebusd /usr/local/bin/ebusd --updatecheck=off -d udp:192.168.7.225:5000 -c /etc/ebusd/wolf -f -r 
2025-07-07 11:15:18.411 [device notice] transport opened
2025-07-07 11:15:18.411 [main notice] ebusd 25.1.25.1 started on device: udp:192.168.7.225:5000, UDP, readonly
2025-07-07 11:15:18.422 [bus notice] bus started with own address 31/36
2025-07-07 11:15:18.425 [bus notice] signal acquired
2025-07-07 11:15:18.425 [bus error] push poll message: ERR: generic device error
2025-07-07 11:15:20.459 [bus notice] new master 30, master count 1
2025-07-07 11:15:20.464 [bus notice] new master 03, master count 2
2025-07-07 11:15:20.464 [update notice] received read betrd_bm2 bm2 QQ=30: aus;Verbraucheraus;5.00;-;-;10.0;8
2025-07-07 11:15:21.816 [update notice] received update-read broadcast betrd QQ=03: 0;00;-;24.5;24;48;20
2025-07-07 11:15:22.875 [update notice] received unknown MS cmd: 305050140720000500194014 / 09000000800080008828
2025-07-07 11:15:23.881 [bus notice] new master f1, master count 3
2025-07-07 11:15:23.881 [update notice] received update-read broadcast betrd QQ=f1: 0;00;-;24.5;-;48;18
2025-07-07 11:15:24.060 [bus error] push poll message: ERR: generic device error
2025-07-07 11:15:25.058 [bus notice] new master 70, master count 4
2025-07-07 11:15:25.145 [update notice] received read mm status_mm QQ=70: 00;21.602;00;20.844;5a;0a
2025-07-07 11:15:25.473 [update notice] received read cha kesselsolltemperatur QQ=30: 5.0
2025-07-07 11:15:25.801 [update notice] received read betrd_bm2 temp_return QQ=30: 24.0
2025-07-07 11:15:26.216 [update notice] received read betrd_bm2 temp_return QQ=30: 24.0
2025-07-07 11:15:26.630 [update notice] received read betrd_bm2 temp_return QQ=30: 24.0
2025-07-07 11:15:26.959 [update notice] received read betrd_bm2 temp_return QQ=30: 24.0
2025-07-07 11:15:27.373 [update notice] received read betrd_bm2 temp_return QQ=30: 24.0
2025-07-07 11:15:27.701 [update notice] received unknown MS cmd: 3008502203cc6c01 / 020000
2025-07-07 11:15:28.115 [update notice] received unknown MS cmd: 3008502203cc6c01 / 020000
2025-07-07 11:15:28.423 [main notice] found messages: 528 (0 conditional on 0 conditions, 6 poll, 124 update)
2025-07-07 11:15:28.529 [update notice] received unknown MS cmd: 3008502203cc6c01 / 020000
2025-07-07 11:15:28.858 [update notice] received unknown MS cmd: 3008502203cc6c01 / 020000
2025-07-07 11:15:29.272 [update notice] received unknown MS cmd: 3008502203cc6c01 / 020000
2025-07-07 11:15:29.687 [update notice] received read cha sammlertemperatur QQ=30: -
2025-07-07 11:15:30.033 [bus error] push poll message: ERR: generic device error

root@tb /h/hc# sudo -u ebusd /usr/local/bin/ebusd --updatecheck=off -d udp:192.168.7.225:5000 -c /etc/ebusd/wolf -f -s
2025-07-07 11:15:39.429 [device notice] transport opened
2025-07-07 11:15:39.429 [main notice] ebusd 25.1.25.1 started with broadcast scan on device: udp:192.168.7.225:5000, UDP
2025-07-07 11:15:39.440 [bus notice] bus started with own address 31/36
2025-07-07 11:15:39.475 [bus notice] signal acquired
2025-07-07 11:15:39.561 [bus error] arbitration start error
2025-07-07 11:15:39.646 [bus error] arbitration start error
2025-07-07 11:15:39.707 [bus notice] new master f1, master count 2
2025-07-07 11:15:39.708 [update notice] received update-read broadcast sollw QQ=f1: 5.000;18.500;0;40;10.000
2025-07-07 11:15:39.714 [bus error] arbitration start error
2025-07-07 11:15:39.800 [bus error] arbitration start error
2025-07-07 11:15:39.800 [bus error] poll mm mischermotor_offen failed: ERR: arbitration lost
2025-07-07 11:15:40.113 [bus notice] new master 70, master count 3
2025-07-07 11:15:40.199 [update notice] received read mm status_mm QQ=70: 00;21.602;00;20.844;5a;0a
2025-07-07 11:15:40.500 [bus notice] new master 30, master count 4
2025-07-07 11:15:40.528 [bus notice] new master 03, master count 5
2025-07-07 11:15:40.528 [update notice] received read betrd_bm2 performance_pump QQ=30: 0
2025-07-07 11:15:40.856 [update notice] received read betrd_bm2 no_of_power_on QQ=30: 71
2025-07-07 11:15:41.270 [update notice] received unknown MS cmd: 3008502203cc362a / 020080
2025-07-07 11:15:41.683 [update notice] received unknown MS cmd: 3008502203cc372a / 020080
2025-07-07 11:15:41.834 [update notice] received update-read broadcast betrd QQ=03: 0;00;-;24.5;24;48;20
2025-07-07 11:15:42.076 [update notice] received unknown MS cmd: 3008502203cc7c2a / 020080
2025-07-07 11:15:42.490 [update notice] received unknown MS cmd: 3008502203cc7c2a / 020080
^C2025-07-07 11:15:42.762 [main notice] SIGINT received
2025-07-07 11:15:42.863 [device error] transport closed

maleadt, 2025-09-26:
Maybe related: https://github.com/john30/ebusd/discussions/1516

hce, 2025-09-27:
Time to do my own ebus implementation...