john30/ebusd #1591: bai various errors from new install

gwaitsi, 2025-09-24:
### Description

In MQTT, I see the following devices;

    ebusd Broadcast (outdoor temp sensor by the looks of it)
    ebusd bai (appears to be the ecotec plus)
    ebusd 470 (appears to be vrs560)
    ebusd (shows the running, scan, signal, uptime status) - also shows update available even though i have 25.1

I see the below errors in the logs.

off topic, is ebusd bai WaterpressureMeasureCounter - the water pressure? It shows as 166 
but in Germany it is Bar. But i don't find any other water pressure option 

p.s. in Home Assistant, it is showing there is an update available, even though it is the latest version 25.1

### Actual behavior

various errors in the logs after new install

### Expected behavior

I assume this isn't a VortexFlowSensor for example. If the device is not detected, it shouldn't produce an error. 

### ebusd version

25.1

### ebusd arguments

Home Assistant 25.1 enh:IP:3333 Scan = on

### Operating system

other

### CPU architecture

x64

### Dockerized

None

### Hardware interface

other

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

[update error] unable to parse poll-read bai VortexFlowSensor from 3108b509030dd500 / 00: ERR: invalid position
[mqtt error] decode bai VortexFlowSensor: ERR: invalid position
[update error] unable to parse poll-read bai Status16 from 3108b5040116 / 00: ERR: invalid position
[update error] unable to parse poll-read bai Status from 3108b5110103 / 00: ERR: invalid position
[bus error] poll 470 CcTimer_Saturday failed: ERR: wrong symbol received
[bus debug] ERR: wrong symbol received during send command, switching to skip
[mqtt debug] publish ebusd/bai/Status01 {
"temp": {"value": 35.0},
"temp_1": {"value": 35.0},
"temp_2": {"value": 12.875},
"temp_3": {"value": null},
"temp_4": {"value": null},
"pumpstate": {"value": "off"}}
[update info] received MS cmd: 1008b5100305ff01 / 00
[update notice] received unknown MS cmd: 1008b5100305ff01 / 00

Miq1, 2025-09-27:
Sorry, I mistook the mentioned German "bar" unit to think you are German. 😉 

The "...Counter" suffix could imply number of pressure measurements taken. I am looking for the water pressure data myself in my heat pump installation - with no luck so far.

john30, 2025-09-27:
not an ebusd issue but rather https://github.com/john30/ebusd-configuration/, so please address it there.
an empty reponse (with length byte 0 at the beginning) is always a hint that the message is not understood by the addressee, so looks like an not-100%-compatible definition for your device.
if you post such things in ebusd-configuration, please be sure to include the output of "scan result", as otherwise nobody can check

closed 2025-09-27 by john30